davidmpye/V10_Dyson_BMS #1: The flashing method does not work

jayshin, 2023-08-22:
Hi David,

I've tried the flashing method (Flashing with a Raspberry Pi as the programmer) using the "2023-08-19-flashpi-lite.img" image but nothing happened. Here are the steps I tried:

1. Wrote the .img file to an SD Card on Mac with Etcher
2. Inserted the MicroSD card into the Pi
3. Woke up the battery pack by pushing the little black 'lever' on the top
4. Powered up the Pi - it asked me to set up the keyboard and add a user account
5. Got the bash prompt but nothing happened

Questions:
1. Do I need to run any command here?
2. For the second method, Flash with SWD capable programmer, I need to change the pi os to Windows because the Microchip Studio supports only Windows, right?

davidmpye, 2023-08-22:
Hi Jay, 

To answer your questions

1) No, it should have run automatically on boot, but either it didn't run, or didn't work!

2) No, you cannot use that option unless you have a hardware programmer - it is not for use with a Pi.  Unless you have a JLink or Atmel-ICE or other hardware programmer, you cannot program this way.

I will make a new image shortly with more debug info so I can try to help better - it will write a logfile to the /boot partition, that you can view from a WIndows PC and then attach to this issue, so I can try to help better.

Could you provide a couple of photos showing the connections between the Pi and the pack, as that might be a common reason why it isnt working?

Thanks!

davidmpye, 2023-08-22:
Hi Jay,

Found several errors with the flash image, which meant it would not work.  I had tested them but somehow.....!

Please could you try with the updated image on the github Releases section?  It will create a file called flash_log.txt in the /boot partition - if it does not work, could you paste that here please?

Thanks again

David

jayshin, 2023-08-22:
Hi David,

Thank you for the reply.
Here are the connection photos:

Thanks,
Jay
[image]
[image]

davidmpye, 2023-08-22:
> Hi Jay,
> I'm just trying to get the updated image to upload to github now - give me
> 30 mins :-)
>
> David

OK, success - the image is now on the releases page.

Cheers,

David

jayshin, 2023-08-22:
Hi David,

Seems it still does not work. Here is the log file:
[flash_log.txt](https://github.com/davidmpye/V10_Dyson_BMS/files/12410703/flash_log.txt)

Thanks,
Jay

davidmpye, 2023-08-22:
hi,

That's an error relating to connection between the pack and the Pi.

If you haven't removed the varnish from the pads on the pack, then it won't make contact - they are covered in an insulating varnish, especially inside the holes.

That's why I suggest in the top of the flashing guide that you need to lightly sand them with a fibreglass pencil or very fine sandpaper to remove the lacquer, then ideally solder wires to the pads to make a reliable contact.

David

jayshin, 2023-08-22:
Okay, let me try it tomorrow. Thanks!

jayshin, 2023-08-23:
Hi David,

I removed the varnish from the pads on the pack and soldered jumper pins as in the following photo.
[image]
However, I got the same error.
[flash_log.txt](https://github.com/davidmpye/V10_Dyson_BMS/files/12415504/flash_log.txt)

What else I can try?

Thanks,
Jay

davidmpye, 2023-08-23:
Hi,

Another thought - try waking the pack up by pressing the black lever as you
have been, but a bit after starting the pi.

Maybe give the pi 30 seconds to start booting and then press the button?
It might be worth experimenting a bit with the time delay.

There is a chance that the pack is going back to sleep before the
programming attempt starts.  Is your pack doing the 30 red flash problem
btw?

David

jayshin, 2023-08-23:
Hi David,

Okay, I will try that.
My one blinks red 20 times.

BTW, do I need to flash the image for every retry? Is there any way to run a command after booting?

Thanks,
Jay

davidmpye, 2023-08-23:
At the moment, there's no easy way to do it manually because there's no
username and password set so you can't easily login.

It's something I'll try to remedy.

Another thing to try: keep pressing the black lever while the pi is loading
so the pack stays awake and the red light keeps flashing.  That should make
it possible for the flash software to connect.

David

jayshin, 2023-08-23:
I tried to keep pressing the black lever while the pi was loading. The red blinking was suddenly stopped just before the screen of the selection of a keyboard but the result was the same as before.

davidmpye, 2023-08-23:
Hi,

That is some progress, and indicates that at least gnd and reset are connected properly, as it'll need those to be able to reset the pack and stop the red flashing.  It's worth checking continuity (and connection) between the other pins (SWDIO and SWCLK) and ensuring they are the right way round....

Otherwise I'm a bit short of other ideas!   Has the logfle output changed since you kept pressing the trigger lever?

David

jayshin, 2023-08-23:
I already checked the continuity of all the jumper wires.

The output of the log file was the same as before.

davidmpye, 2023-08-23:
Hi Jay,

the jumper wire connections look ok, and the same as the photos - I can't see the purple wire though - i don't know if the flash of the camera has changed the colours?

davidmpye, 2023-08-23:
Just to check also - is this a pi4?  I'll try to replicate the exact scenario tonight...!

David

davidmpye, 2023-08-23:
OK, I have a couple more suggestions - how comfortable are you with the command line on linux?  If you can add a keyboard and set up a user account, you could replace the rpi2.cfg file in /home/pi with the attached.

It looks like it might need some changes for a rpi4 due to the faster cpu speed.  

[rpi2.cfg.zip](https://github.com/davidmpye/V10_Dyson_BMS/files/12417935/rpi2.cfg.zip)

If you're not so familiar with the linux commandline, I can update the image.

David

jayshin, 2023-08-23:
I'm a software engineer, so no problem with the linux commandline. Could you give me more detailed steps to do it?

BTW, here is a new photo for the wire mappings:
[image]

davidmpye, 2023-08-23:
Hi,

OK, the connections look fine.

If you create a user account, and go into /home/pi, you can edit the file rpi2.cfg using nano (or vi if you prefer).
You will need to use sudo to execute the editor to ensure you have sufficient permissions.

eg:

cd /home/pi
sudo nano rpi2.cfg

The lines to change are as follows:

Change the numbers for bcm2835gpio_speed_coeffs so the line reads:

bcm2835gpio_speed_coeffs 236181 60

Add 2 lines:

bcm2835gpio_srst_num 18
reset_config srst_only srst_push_pull

Then save the file, reboot the pi, and try the flash process again.
Thanks for your patience, we will get there!

David

davidmpye, 2023-08-23:
Oops, and one more line to change!

Sorry!  It seems a few more things have changed on Pi4 than I realised!

Again, in the rpi2.cfg, change the
bcm2835gpio_peripheral_base   address to **0xFE000000**

David

jayshin, 2023-08-23:
Okay, let me try it after dinner.
Thanks!

davidmpye, 2023-08-23:
Sorry about all of this, the idea was to make this image automatic, but it seems that Pi4 has changed quite a few things, and I haven't specifically tested on that model.  Once we've made it work, I'll produce a new automated image which will improve the auto detection!

David

jayshin, 2023-08-23:
Hi David,

I've updated the `rpi2.cfg` under /home/jshin directory (my user id is `jshin`, so there is no directory /home/pi) and rebooted the pi 4 while pushing the black lever but nothing happened.

Do I need to create the user `pi`?

Thanks,
Jay

davidmpye, 2023-08-23:
Hi,

It must have been moved by the system from /home/pi which means the
autoboot system cannot find it.  It may have been done by the system when
you created the user manually.....?

It might be easier to run the flash_dyson script in that directory and see
if that does the flashing.

sudo ./flash_dyson.sh should execute it from within your home dir, with the
pack awake

David

jayshin, 2023-08-23:
Got the same error:

"Error: Error connecting DP: cannot read IDR"
[image]

davidmpye, 2023-08-23:
Could you show me the rpi2.cfg file please?

Eg

cat rpi2.cfg

I am getting a pi4 to test tonight.

Thanks

David

jayshin, 2023-08-23:
Here it is:
[image]

davidmpye, 2023-08-23:
Thanks.

I'll test on a pi4 and get back to you tomorrow with more ideas....
Cheers

David

jayshin, 2023-08-24:
Hi David,

I don't have any older pi, so please let me know if you have anything to try on the pi 4.

I am also interested in hardware programming. I know some programming languages including c/c++. I'd like to hack the BMSs of Dyson, Makita, and iRobot. Where can I start? Any advice?

Thanks,
Jay

jayshin, 2023-08-30:
Hi David,

Have you got a pi 4? Is there any progress on the issue?

Thanks,
Jay

davidmpye, 2023-08-30:
Hi Jay,

I have a Pi 4 and I cannot solve the problem at the moment, despite trying lots of combinations of settings.
So for now, unfortunately, I have had to say it does not work on a Pi4.  Specifically, it is the routine to unlock the microcontroller that doesn't work on a Pi 4. Once unlocked, the microcontroller can be programmed with a Pi4.

For now, I have to suggest getting a Pi Zero or Pi1 etc to borrow temporarily.

Sorry,

David

davidmpye, 2023-09-04:
Hi Jay,

I've now got it working on a Rpi4 (at least, on mine...!)

It turns out it was a timing issue, meaning the chip erase command was not finished before a reset was initiated.

Would you be able to test the updated SD card image when you have a chance please?

David

jayshin, 2023-09-05:
Hi David,

Thank you for the good news!
Unfortunately, my two V10 batteries no longer blink, even when I push the black lever. One of the batteries was 0v, so I replaced the battery with a proper one temporarily but I could not wake it. Do you know how to fix the issue?

Jay

davidmpye, 2023-09-05:
Hi Jay,

It might be related to previous programming attempts -  I did have one that
ended up in this state, but could still be fixed.

The cells will all need to be 'ok' - ie no 0v cells, so replacing any dead
ones is important.   When you either plug the pack in to a charger, or
press the black button, does 3.3v appear on the lowest pin in the picture
below (you can measure between the 3v3 pin and gnd next to it)?  If 3.3v
appears, then the power subsystem is awake, so it will power the
microcontroller, and should be possible to flash it still.

David
[image: image.png]

jayshin, 2023-09-06:
Hi David,

I confirmed that there was 3.3v between the lowest pin and GND and I tried the new one 3 times but failed as shown in the attached log.
[flash_log.txt](https://github.com/davidmpye/V10_Dyson_BMS/files/12532482/flash_log.txt)
Could you please check the log?

Thanks,
Jay

davidmpye, 2023-09-06:
Hi Jay,

That log appears to be the old image, as the new log file should look quite
different.

The new one uses sysfsgpio and not the bcm bitbang driver.

Have you ensured you're using the new updated image I uploaded?

David

davidmpye, 2023-09-06:
Hi Jay,

I've just redownloaded the image, and confirmed it's correct - that log
file is definitely from a run of the old image and not the current one.

David

jayshin, 2023-09-06:
Hi David,

I've redownloaded it and retested it.
[flash_log.txt](https://github.com/davidmpye/V10_Dyson_BMS/files/12534102/flash_log.txt)

and md5 check:
sjay@MacBook-Pro Downloads % md5 rpi-v10-flash.zip 
MD5 (rpi-v10-flash.zip) = ddb1a4b993494c6fa359b720f540eccd

Jay

davidmpye, 2023-09-06:
My bad, that does look like the new file.  I think the caching was this
end, showing me an old version of flash_log.txt!

Had a read, basically it cannot establish any comms with the
microcontroller on the pack.

As far as I can see, assuming the code is OK (and I've tested it quite a
bit here), either:

Pack MCU is dead/damaged
Connection problem between pack and Pi.
One or more gpio pins on the Pi are damaged or not working

Do you get the same result on both packs?   I'd be interested to see if you
have the same problem when the Pi zero arrives....

David

jayshin, 2023-09-06:
Tested it with another battery pack but got the same error:
[flash_log.txt](https://github.com/davidmpye/V10_Dyson_BMS/files/12534956/flash_log.txt)

jayshin, 2023-09-06:
Could you recheck the line connections in the following photos?
[image]
[image]

davidmpye, 2023-09-06:
Hi,

The photos show correct connections.

But a couple of cautions: I can't tell from the photo if those pins have
full continuity to the holes ie if varnish fully removed and good
conductivity.

Also there is a metal strip from the battery that runs close to / under the
pins.  If at any point any pins have shorted to that strip, it may have
damaged the gpio on either the MCU or the pi.

You mentioned neither packs flash red any more - do you know when/how that
happened?

I'm sorry to not be able to offer a definite solution but it's
understandably difficult to debug this kind of thing remotely.

David

jayshin, 2023-09-06:
Hi David,

No worries about the cautions. I carefully connected and tested with my multimeter tester.

I'm not sure when the red flash disappeared. When I attempted to connect the battery to the Pi Zero after the last test with the Pi 4, it didn't work. The other one is functioning well after replacing the battery.

Jay

davidmpye, 2023-09-06:
Hi Jay,

Have you flashed one pack successfully now?

David

davidmpye, 2023-09-06:
> My bad. The other one is not functional but red blicking only. Got the
> same result with that one as I reported earlier.

I see.

I don't think I'm going to be able to figure out anything more with your pi
4 from here, sorry.  It certainly works on the Pi 4 Model B I have tested
on, and I cannot figure out why yours doesn't appear to - sorry.

Do let me know when your Pi zero arrives and whether that fixes the problem.

Best wishes

David

davidmpye, 2023-09-06:
> BTW, what is the default username and password of the new image?

Hi, should be as standard: pi and password raspberry.

David

ChristianRomberg, 2023-09-28:
Hi, reading through this thread and the flash_log.txt files, it seems I had a similar issue to @jayshin. What fixed it for me was playing around with the black button on the battery pack during the flashing, that might have prevented the battery pack to go to sleep.
I had the same errors in my flash_log and that worked for me

davidmpye, 2023-09-29:
Hi Christian

Thanks for your report.

Unfortunately, those log errors are just generic "I couldn't get a
response from the SAMD20" - that's what makes it really frustrating to try
to remotely debug.

It could be a connection issue, it could be RPi GPIO issue, it could be a
power issue, the chip might be fried - and there's no way to tell...!

I'm going to release a new FW version shortly, which fixes a number of
bugs, and properly uses the charge counter so the battery LEDs are more
accurate in representing pack charge.

Cheers

David

jayshin, 2023-10-05:
Hi David,

I just purchased a Pi Zero W v1.1 today. I'd like to try it with the new FW you mentioned above.

Thanks,
Jay

davidmpye, 2023-10-06:
Hi Jay,

If you want the latest 'test' firmware, I've attached a zip file.

Read the 'boot' partition on the Pi's SD card - you should find a file there called V10_BMS.elf (A windows PC should be able to read that partition)

Delete V10_BMS.elf, and copy the one from the attached zip file into its' place, and then eject the SD card from the PC.

Then, do the flashing process to write the new firmware image onto the pack. (Note the pack needs to be 'awake' - so you need to press the trigger button once on the pack when you turn the Pi on and it should stay awake long enough)

[V10_BMS.zip](https://github.com/davidmpye/V10_Dyson_BMS/files/12828916/V10_BMS.zip)

davidmpye, 2023-10-06:
Hi Jay!

Great news, thanks for sticking with it, and I am delighted it's now working for you :-)

Any nnew problems you encounter, please open an Issue !

Regards

David

closed 2023-10-06 by davidmpye