davidmpye/V10_Dyson_BMS #2: Non-Working Battery

puppywang, 2023-09-02:
Thank you very much for your project. I have a potentially faulty V10 battery. The symptoms are as follows: after a full charge, pressing the power button only results in a brief noise, and then the battery light starts to flash. If I recharge it, it starts from the second bar and quickly fills up to three bars. However, pressing the power button again produces the same symptoms. I disassembled the battery and measured the voltage of each cell, which ranged between 4.05 to 4.07 volts, with a total voltage of 28.4 volts. I successfully unlocked the security lock using OpenOCD and flashed the V10_BMS.elf file from your project. After flashing, pressing the power button doesn't turn off the battery light, but strangely, the vacuum cleaner doesn't work at all. I measured the output voltage of the battery; it's 0.1V when the switch is not pressed and 28.41V when pressed. I also checked the serial port line interface, and all lines are connected. I ruled out the issue with the vacuum cleaner itself, as I ordered a new original battery and the vacuum cleaner resumed working. However, I'm completely stumped with this old battery, as if I've hit a dead end. Do you have any suggestions for me? Thank you.

davidmpye, 2023-09-02:
Hi,

This is really interesting - thank you for trying out my firmware.

On the original firmware, the pack will shut down if it doesn't receive
serial communication from the vacuum on startup, whereas mine deliberately
doesn't (in case you wanted to use the pack with something else?).  If ~28v
is appearing when you push the button, the firmware is enabling the output
because the pack has passed the safety checks. (so far so good!)

If the vacuum won't start despite volts being present, I think it most
likely relates to the serial communications with the cleaner.   If you have
access to an oscilloscope, you can see what is going on.   With my
firmware, if you probe between 0v and the serial lines (you'll have to try
both,I can't remember which one is tx and which is rx!)  you should see
outgoing data (which is what the cleaner needs to see in order to run). My
firmware doesn't care about communication from the cleaner so you don't
need to worry about data in that direction, but the cleaner definitely does
care about receiving data from the pack, else it won't start.

David

puppywang, 2023-09-03:
> 

I don't got one oscilloscope, but I have some home made logic analyzer, I will try to sniff the commuication between them.

puppywang, 2023-09-04:
@davidmpye  The voltage on TX/RX line is around 25.81v, it is far beyond my logic anlayzer's voltage input range (0~5v). Do you know another TX/RX pin that have 0~5V ttl level ? Thanks.

davidmpye, 2023-09-04:
Hi,

Yes I do.  On the picture attache, you can see two ceramic capacitors labelled c13 and c43.

[image]

On the sides of them nearest the camera, you should find the signals at 3v3 levels.  But do check first!

David

davidmpye, 2023-09-04:
Sorry, should have added: they carry 115200 baud serial data.

puppywang, 2023-09-04:
> 

Very appreicated, I will try later.

puppywang, 2023-09-18:
@davidmpye Sorry for the delay, I've trace the data sent from a working battery, the data is different everytime, might dyson upgrade their protocol ? I've collect some trace and attach to this issue, do you have any advice that I can follow? Thanks.

[4.log](https://github.com/davidmpye/V10_Dyson_BMS/files/12644833/4.log)
[5.log](https://github.com/davidmpye/V10_Dyson_BMS/files/12644834/5.log)
[3.log](https://github.com/davidmpye/V10_Dyson_BMS/files/12644835/3.log)

davidmpye, 2023-09-18:
Hi!

This is really interesting - can I ask a couple of questions?  Are these
dumps 1 or 2 way, and how can I tell what is coming from battery to cleaner
and viceversa?

Is this the same V10 battery type shown in the photo of the project github
page, with the series of leds down the side?

Does the battery you flashed with my firmware output any serial data over
the link-connection to the cleaner?  If so, could you capture it - and even
better, the 2-way traffic showing what the battery sends, and what the
cleaner replies?

It is possible the protocol was updated - the basics seem similar but there
are some differences in the message types, which might explain why the
cleaner won't run....

David

puppywang, 2023-09-19:
Sure, I am glad to share any information I have.

1. I found two version of Dyson V10 battery here, the 2300mAH and 2600mAH version, the 2300mAh one is equip with Molicell ceil and the 2600mAh have the SONY ceil, the data is captured from the 2300mAH version, and is the data I capture is from battery to motor.

[image]
[image]

2. I flash V10 firmware in this repo to another battery, the same 2300mAH battery, without luck, the moto won't start, I will capture and attach the commuication between them later.

3. My old logic analyzer is not working anymore, so I use the USB-UART converter to capture the traffic, so the traffic is one way, I will try to capture using two USB-UART converter later.

By the way, I found there is a serial-debug function in your new V10 firmware, and the PIN you are using is PA10 and PA11, is this two pin available for connect from the TEST PIN ? I want to check the log from the battery, Thanks.

davidmpye, 2023-09-20:
Hi,

Yes, you're right - the serial debug function uses the two pins on the test
connector  - the pins at the end of that test connector go

RESET
PA31 SWDIO
PA30 SWCLK
PA00
PA11
PA10
GND
Vdd (3.3v)

So PA10 and PA11 can be connected to a USB-serial converter or similar at
115200baud and you can see the debug output. NB you'll have to compile the
firmware and enable USB DEBUG in config.h for it to work.

The serial data my firmware sends is a short series I captured from a
'clone' Dyson pack and drives my cleaner just fine.  I'm curious as to why
it doesn't seem to drive yours, so if we can check that the firmware is
correctly sending it to the cleaner via your 2x USB-UARTs, and what the
cleaner sends back, it might give us an idea what is wrong...

I'm not sure the debug output from PA10/11 will be that helpful but it
would be interesting to see what it says.

David

davidmpye, 2023-09-21:
Hi,

I've tried making a firmware with the debug data that your cleaner sends, just to see what happens!
On my cleaner, it does not start.  I wonder if your cleaner will run on it?

It seems there are 3 generations of 'V10 Dyson' - do you know what kind yours is?
Mine is an SV12 (it's on the bottom of the machine, but you have to remove battery to see it.) I wonder if the protocols are different for later editions?

David

[V10_BMS.zip](https://github.com/davidmpye/V10_Dyson_BMS/files/12693181/V10_BMS.zip)

puppywang, 2023-11-29:
@davidmpye Sorry for the delay, I am unable to test the firmware right now, as the battery died when I connect my USB-TTL to the PA10/PA11 debug pin. It start printing in the beginning, than I connect the battery to the vacuum to see what happened if I press the start button, then the battery died, the voltage on the VCC is always 0.3V, and the TI chip is a little bit hot, so I am unable to test the firmware anymore. Sorry for that.

But the way, I've double check the version of my V10 vacuum, at the place you metioned, It marked SV12 Animal version, that a different version than yours? And I've collect full log of a working battery, Could you check if the protocol inside the new log will able to power your vacuum, thanks.

[dump_no_filter.log](https://github.com/davidmpye/V10_Dyson_BMS/files/13504564/dump_no_filter.log)
[dump_success_power_up.log](https://github.com/davidmpye/V10_Dyson_BMS/files/13504566/dump_success_power_up.log)
[dump_fail_power_up.log](https://github.com/davidmpye/V10_Dyson_BMS/files/13504567/dump_fail_power_up.log)

turenabo, 2024-06-09:
> > 
> 
> I don't got one oscilloscope, but I have some home made logic analyzer, I will try to sniff the commuication between them.

hi, puppy, can you share some information about your logic analyzer? I want to get one too. thanks