davidmpye/V10_Dyson_BMS #3: Battery not charging after flashing successfully!

viostn88, 2023-10-04:
Hi David,

I have two Dyson V10 batteries with the same behavior: red light flashing, not charging. I've tried the flashing method with a Raspberry Pi 2, Model B, v1.1 as the programmer. The file used was v1.0.1, date of the file 9/5/2023. The success was 50%.
The first one (bought from Dyson as a spare part) was recovered, after flashing the firmware is working well. 
The second one (that came with the vacuum cleaner) is not working after flashing the firmware, even if the messaje was the operation completed successfully. After flashing, the pack will do the light sequence but when I press the lever the first light is flashes blue very quickly 4 or 5 times (hard to count) When I connect the charger this light is flashing red 3 times, the same as before flashing the firmware. Disconecting the charger and pressing the lever the pack have the same behavior, will do the light sequence and flashes blue very quickly, connecting again the charger the red light flasing again 3 times and no charge.
All cells are tested, the same voltage 3.8V, and seems to be good.
The first battery, recovered, is made in Dec. 2022 and the one that came with the vacuum cleaner is made in May 2018.
Do you think that the second battery controller is defective? Or maybe the hardware is different and the firmware is not working with it?
Thank you!

davidmpye, 2023-10-05:
Hi!

Glad to hear the first pack is working OK!

For the second pack:

If you get the full light sequence  (ie all the leds go on then off in sequence) when you 'wake the pack' from sleep mode (ie press the trigger, after leaving it for 15 mins) then the new firmware is programmed on properly.   If it doesn't do the 'light show' on power up, then it might not, and reflashing would be worth a go.

If the firmware is flashed correctly, and you are trying to charge and get three red flashes, that indicates the BMS is reporting BMS_ERR_CELL_FAIL.  This will occur if any of the cells are read as being <2.0V by the controller.

I do wonder if despite your multimeter readings, you do have a faulty cell, but it's hard to be certain from here.

If you have access to a USB-serial converter, I can tell you how to get more debug info out of the pack - let me know.

Best wishes,

David

viostn88, 2023-10-05:
Hi,
Thanks for your fast reply. 
I checked all the cell on the second pack and have 3.8 V each. No one is defective, or discharged, that not the case.
The firmware was programmed on properly and do the "light show" when I press the lever. But it lights up blue quickly several times after pressing lever compared with the first pack which light blue once.
I have a USB-serial converter and I would be happy if you could give me more indications to explore further on the serial and get more debug info from the pack.
All the best for you!

Stan

davidmpye, 2023-10-05:
Hi Stan,

Great news.

Grab the attached file and unzip it.

[V10_BMS.zip](https://github.com/davidmpye/V10_Dyson_BMS/files/12822671/V10_BMS.zip)

Read the 'boot' partition on the Pi's SD card - you should find a file there called V10_BMS.elf  (A windows PC should be able to read that partition)

Delete V10_BMS.elf, and copy the one from the attached zip file into its' place, and then eject the SD card from the PC.

Then, repeat the flashing process to write the new firmware image onto the pack. (Note the pack needs to be 'awake' - so you need to press the trigger button once on the pack when you turn the Pi on, and it should stay awake long enough)

To read the debug output, you'll need to connect your USB-serial device to the following pins on the debug header:

[image]

Connect GND of the USB-serial connector to the GND pin on the header.
Connect the RX pin of the USB-serial connector to the pin immediately above that (unlabelled)

You don't need to connect TX, or 3v3.  Just those 2 will do.
The settings are 115200 baud.

Let me know what you see! :-)

viostn88, 2023-10-07:
Hi David,
I followed the guide you sent.
The serial log is attached. Seems that the 6th cell is not OK?
With my multimeter all cells have the same voltage. Replacing the 6th cell and flash the firmware again nothing positive happend. Maybe a controller failure!
Do you think there is anything else to try?
Thank you,

Stan

[DysonV10.txt](https://github.com/davidmpye/V10_Dyson_BMS/files/128374
[image]
85/DysonV10.txt)

davidmpye, 2023-10-08:
Hi Stan,

Looks like either the BMS chip itself is faulty, or there is a fault with
the connection between it and that cell.

If you look closely near each battery pin, you'll see a pair of tiny SMD
resistors, connected to each battery pin by a fine track.
I'd check those resistors are in place (I've occasionally damaged them when
opening packs) and that they aren't open circuit (for some reason?)

Other than that, I cannot think of an easy fix to the problem, sorry

David

viostn88, 2023-10-08:
Hi David,

Already checked the pair of resistors 1K and 5K with my multimeter and they are good. I cannot see damage of the tracks.
I assume that the controller is faulty. I will reuse the cells.
Thank you very much for your work and support for battery recover.
Best regards,
Stan

closed 2023-10-08 by davidmpye

viostn88, 2023-11-01:
Hi David,
I'm back with an update.
Finally I think that I found the cause of the error on the second battery (Battery2).
I focused on cell 6 but later I realized that the counter starts from zero and the problem was actually from cell 7!!!
I suppose the Rf resistor is broken because the measured voltage on BAT was 25V and the PACK+ was 27.2V. I replaced it with one of 100 ohm. Now the battery is charging and working but I am not sure that the capacity is like before giving the error. The vacuum cleaner was used only for 15 min. I will keep testing to see if it works more than that. Also only first light is flashing when charging, the the others do not light up even if it stays on charger all the time.
Attached you will find the schematic diagram with the bq76930 and the photos of the labels of the two batteries.  Battery1 is the new one bought from Dyson shop in 2022 and the Battery 2 is one who came with the vacuum cleaner (bought in 2018). Maybe it is useful for you.
Best regards,
Stan
[Dyson battery.pdf](https://github.com/davidmpye/V10_Dyson_BMS/files/13231320/Dyson.battery.pdf)
[image]
[image]

viostn88, 2023-11-01:
Schematic diagram
[image]

davidmpye, 2023-11-19:
Hi!

That's really helpful, thank you for adding it!

 It doesn't sound like the pack full voltage is being registered correctly, which means charging isn't proceeding properly, and discharging is being stopped earlier than it should be....

If it's helpful, I can measure the original resistor value if you aren't sure what it should be.   Can you show me where it is on the PCB to save me the job of reverse engineering it all? ;-)

David

viostn88, 2023-12-05:
Hi David,
On the BQ76930 datasheet they say that the Rf resistor must have min 40, typ 100, max 1000 Ohms. I don't know where is Rf on the board. Need to remove the silicon that cover the whole board to find it. But now I don't have time to do that. Maybe later this month.
Thanks for help, I will be back when I have some news.
Viorel

s-ata, 2024-06-11:
Hi, I just pass by to say I have the same issue here. 

I can manually measure all cells are on 3.2V +-30mV, but the battery has 3 red blinks (first led only) when I attempt to charge it, indicating the reading of one cell is below 2V as the wiki says. I would say the firmware was flashed correctly. For reference, I have the V12 version of the battery too, with sony cells. It was doing the ~30 red blink before flashing.

I will try later in the week to remove the glue over the pcb and measure these resistors, keep you updated and upload a photo of that RF resistor if I manage to find it. Hopefully I can savage the board because I already bought new cells from nkon :(

I would like to take the chance to also say thanks for the nice project David!

davidmpye, 2024-06-11:
If you wish more debugging, the pin header on the pack has a serial debug
output which will tell you which cell and what the reading is

David

s-ata, 2024-06-11:
I was hopping to do it quick, but my electronics fall short here and I would like to avoid removing the cells if possible. Is there any chance you remember the placement of these RF resistors?

[image]

At the moment I don't have the tools to connect to the serial port easily unfortunately

davidmpye, 2024-06-11:
I don't know, sorry.

You could even use a Pi to connect to the serial debug UART I should think.

You would need to use raspi-config to enable Serial UART on the pin header,
and use screen or minicom or a similar program to open it.  The debug port
runs at 115200baud.

David

s-ata, 2024-07-03:
I gave it a second go today, but unfortunately I put 13V on one of the data pins while trying to debug it. Me and my big hands....

A reminder to the following person, be aware when soldering, the pins, as they are just above the cell strip, if they get in contact you will summon the magic smoke on the BMS!

Just put some tape to be safe, my novice advice.

requiem182, 2024-08-08:
I have the same issue unfortunately.

requiem182, 2024-08-08:
I have the same issue unfortunately.  Blue lights will flash when button is pressed but won't charge. Getting 3 red blinks repeatedly.

davidmpye, 2024-08-09:
You've either got a flat cell (that needs fixing) or a hardware fault that
is preventing the BMS from reading the correct cell voltage.

You need to find and fix the hardware fault before you can safely expect
the pack to work

requiem182, 2024-08-09:
> You've either got a flat cell (that needs fixing) or a hardware fault that is preventing the BMS from reading the correct cell voltage. You need to find and fix the hardware fault before you can safely expect the pack to work
> […](#)

I found out that the cells was between 1.75 - 2v. I did took it apart and charger it, all cells still reach 4.2v. My problem now is I didn’t took a photo of battery orientation.😅