davidmpye/V10_Dyson_BMS #4: Cell Balancing

aaronsb, 2023-12-02:
I think it still makes sense to try and attempt cell balancing even with the constraints placed on the BMS board design. In many real world scenarios, it could be days to a week or so between uses. Any balancing attempts within the safe current limits should be beneficial.

Additionally, units stored without use for longer intervals benefit more, due to the likelihood of high power intermittent usage are likely to unbalance the pack, and therefore a need for cell balancing even more. The extra time available stacks up the likelihood of successful balancing.

davidmpye, 2023-12-27:
Hi Aaron,

Thanks for your comments.   I think, overall, I agree.  It would need a new state adding to the BMS state machine - ie BALANCING.  However, I don't think I'll have a chance to do this in the foreseeable future.

If you fancied having a go, and did make a patch, I'd be happy to review and merge it :-)

David

Mugrunty, 2025-01-23:
I started playing around with some basic balancing code on my failed crap V10 battery.  If I ever get it working to a decent degree, I'd be happy to make a patch!  It's too crude right now to put it in, but I have a basic algorithm that balances the pack after it finishes charging and is in the state PLUGGED_IN_NOT_CHARGING.  It'll keep balancing until the highest cell is no longer the highest or when you unplug it.  But it needs some polishing and other logic for when there's only a single LOW cell.  The ESR has risen to a nearly unusable level in my vacuum, but I can keep playing around with it and eventually put it back together for fun. :)