davidmpye/V10_Dyson_BMS #7: Commands to bypass protection

TorxPacman, 2024-01-18:
Hi.  I'm stuck trying to program ATSAMD20E15 with a Black Pill (Black Magic Probe fimware) but I can't because of the protection bits. Can you tell me which commands you used to connect  with SWD interface (bypassing the protection and erasing the flash, i.e. the ones you used to make the file for Raspberry PI).    Thanks

davidmpye, 2024-01-22:
Hi,

Sure - have a look here:

https://github.com/davidmpye/pi-v10-dyson-flash/blob/master/stage2/04-dysonfiles/files/flash_dyson_v10.sh

The key thing is that first of all you have to bring up the MCU in 'coldplug' mode, by doing the following dance with the SWCLK and SRST pins:

```

	echo "Exporting SWCLK and SRST pins."
	echo $SWCLK > /sys/class/gpio/export
	echo $SRST > /sys/class/gpio/export
	echo "out" > /sys/class/gpio/gpio$SWCLK/direction
	echo "out" > /sys/class/gpio/gpio$SRST/direction

	echo "Setting SWCLK low and pulsing SRST."
	echo "0" > /sys/class/gpio/gpio$SWCLK/value
	echo "0" > /sys/class/gpio/gpio$SRST/value
	sleep 1
	echo "1" > /sys/class/gpio/gpio$SRST/value

	echo "Unexporting SWCLK and SRST pins."
	echo $SWCLK > /sys/class/gpio/unexport
	echo $SRST > /sys/class/gpio/unexport

```

Once you've done that, you can do the actual unlock.
I do it like this:

echo "Unlocking processor"```

./openocd --debug -c "adapter driver sysfsgpio; \\
sysfsgpio_swclk_num $SWCLK; \\
sysfsgpio_swdio_num $SWDIO; \\
sysfsgpio_srst_num $SRST; \\
transport select swd; \\
source [find at91samdXX.cfg]; \\
reset_config srst_only; \\
init; at91samd.cpu mwb 0x41002100 0x10; sleep 500; reset; halt; exit"\
```

David

davidmpye, 2024-01-22:
Alternatively, if Atmel Studio will work with a Black Pill, you can just use the unlock feature in that.

David

closed 2024-04-16 by davidmpye