davidmpye/V10_Dyson_BMS #9: There is a little delay compared to the original battery

turenabo, 2024-04-14:
hi David, many thanks for this project I resurrected my old battery, but it's a little delay compared to the original battery. with original battry, it's almost same time the motor started when I push the button, but it will delay a little bit when I try your firmware. Do you know why?

davidmpye, 2024-04-16:
Hi,

There is a longer delay on my pack when it wakes from sleep - so, the first time you pull the trigger, but if should be faster after that.

If that is what you noticed, could you try the new release: 

https://github.com/davidmpye/V10_Dyson_BMS/releases/tag/v1.2

and see if that fixes?

Thanks!

turenabo, 2024-04-17:
Thank you very much, I would be really happy if you released a new version because of the issue I submitted. 

Yes, there was indeed a longer delay, a few times I had already pushed the vacuum cleaner out and the motor hadn't started yet. It’s not just this delay. Compared with the original firmware, the original firmware starts the motor almost immediately when your finger presses the button. Your firmware has some delays compared to the original firmware.

I will try your new release today. 

btw, you mentioned that there is a cell imbalance led code, how many mv does it flash once?

davidmpye, 2024-04-17:
Hi,

From:

https://github.com/davidmpye/V10_Dyson_BMS/wiki/New-firmware-operation

*If the charger is disconnected while charging is in progress, a series of
brief red flashes will show if the pack is out of balance. If no red
flashes show, the pack is well balanced, with highest and lowest cell
voltage within 50mV of each other. Each red flash (if shown) indicates a
50mV difference between highest and lowest cell voltages.*
Cheers

David

turenabo, 2024-04-21:
> Hi,
> 
> There is a longer delay on my pack when it wakes from sleep - so, the first time you pull the trigger, but if should be faster after that.
> 
> If that is what you noticed, could you try the new release:
> 
> https://github.com/davidmpye/V10_Dyson_BMS/releases/tag/v1.2
> 
> and see if that fixes?
> 
> Thanks!

Hi David, I just tried v1.2, let me report to you what I found. It becomes faster than before, but it still need take longer time to wakeup, I compared it again with the original firmware, seems the original one no need wakeup, even if I leave it for a long time it can start immediately, but yours need wate. And it's good is once both of them have woken up, there is no difference now.

turenabo, 2024-06-11:
> Hi,
> 
> There is a longer delay on my pack when it wakes from sleep - so, the first time you pull the trigger, but if should be faster after that.
> 
> If that is what you noticed, could you try the new release:
> 
> https://github.com/davidmpye/V10_Dyson_BMS/releases/tag/v1.2
> 
> and see if that fixes?
> 
> Thanks!

Hi David, I still don't like the delay when the battery wakes up, so I tried removing the LED sequence function, and then there is no delay. I looked at your code and it seems that there is no timer used. If you use a timer to make a non-blocking LED lighting function, there should be no delay, right?

davidmpye, 2025-01-01:
That would be correct - pull request welcome :-)

closed 2025-01-01 by davidmpye