davidmpye/V10_Dyson_BMS #10: Instruction

kulic23, 2024-04-17:
Hi David, could you write a detailed instruction on how to compile a firmware file from the source file, for example, if I use high-capacity cells, I understand it also needs to be indicated in the code, I am a specialist in battery assembly but I do not understand anything about programming, I need instructions on how to change the battery capacity in the code and compile a new firmware, thank you very much

turenabo, 2024-04-17:
seems david's firmware manages cells charge and discharge by voltage

davidmpye, 2024-04-17:
Hi,

Yes that is correct.  It does use the charge counter capability to learn
the cell capacity to gauge battery capacity but it will do that
automatically.

The cut offs for full and empty are voltage based and in the config.h file.

You can build the firmware if you need to change these values yourself
using Microchip Studio (download link on wiki)

It's important to have high discharge current ratings for the cells as they
do get a hard life....

David

turenabo, 2024-04-18:
> Hi, Yes that is correct. It does use the charge counter capability to learn the cell capacity to gauge battery capacity but it will do that automatically. The cut offs for full and empty are voltage [...]

Hi David, where can I find a brief of "charge counter capability to learn the cell capacity"? I am learning BMS, it's an interesting topic. I see you mentioned your firmware doen't have coulomb meter, is this what you are talking?

davidmpye, 2024-04-18:
Hi,

Yes it does use the coulomb counter - I need to update the wiki page to
explain that.

The CC is used to estimate the pack capacity by observing charge flow on
charge and discharge.

Have a look on the wiki - there are links to the BMS IC datasheet which
might help give you some ideas.

Cheers

David

closed 2025-01-04 by davidmpye