davidmpye/V10_Dyson_BMS #11: Works great.  Programmed with Pi 4

MrEdmondSiu, 2024-05-09:
Hi David, 

I just wanted say thanks, I was able to flash the FW using pi 4 and the instructions and got a locked out V10 battery working again.

The only caveat was that I needed to connect the 3.3V on the pi to the battery for the flash to work.  Your instructions in the wiki says 3.3V isn't needed when flashing with your img  for the pi.

Thanks again.

snke95, 2024-07-28:
ciao, ho un pi3, ho collegato anche i 3.3v ma mi da errore unlock. mi aiutate ho 3 batterie ferme. grazie

davidmpye, 2024-07-28:
The problem is almost always poor connection - you need to remove the clear
varnish and then solder wires or pins to the pads.  Just pressing or
touching the pads will not work.

snke95, 2024-07-28:
ho saldati i fil perfettamente alla scheda, ma nulla da fare, il pi3 cerca di flashare la batteria si illumina di rosso e da errore "unable to unlock pack"

davidmpye, 2024-07-28:
Are you sure the connections are also correct on the pi?

The Pi usually seems to work eventually - a proper programmer is always the
best solution though....

David

snke95, 2024-07-28:
ho seguito con precisione le istruzioni.....

closed 2025-01-01 by davidmpye