davidmpye/V10_Dyson_BMS #12: How to create a Raspberry Pi firmware burning file？

turenabo, 2024-06-06:
Hi Dave, your project is great. I have unlocked my V10 battery before. I want to make some adjustments to your code recently, but I don't have a programer. I only have a Raspberry Pi. How can I make a Raspberry Pi firmware burning file?  Could you share some info or idea, many thanks.

turenabo, 2024-06-08:
> Hi, Thank you - I am glad you like the project. Use Microchip Studio to build the firmware image using your new changes. Find the .elf file that it built. Insert the Pi's SD/MicroSD card into your PC, and look in the /boot partition -you should be able to do this even on a Windows PC. In the boot partition, you'll find a file called V10_BMS.elf - you need to delete/rename it, and place your generated elf file into /boot, and rename it to "V10_BMS.elf" - the capital letters are important. Then, you can eject the SD/MicroSD from the PC, and use it in the pi again, and it will attempt to flash that image. Kind regards, David
> […](#)

many thanks, I will try

probonopd, 2024-12-30:
https://www.eevblog.com/forum/reviews/dyson-v7-trigger-cordless-vacuum-teardown-of-battery-pack/msg5426312/#msg5426312  says that "flashing back the `00` and `7F` to those two locations" (see the screenshot there) may unlock a locked battery. Is it possible to do that with the software for Raspberry Pi rather than a PICkit?

davidmpye, 2024-12-30:
> screenshot there) may unlock a locked battery with factory firmware.

Good question.

It seems not without separate circuitry.

It might be able to be done with a circuit like this:

https://www.techmeology.co.uk/rpipic/

But I don't know, and unfortunately I don't have any pic development
experience.

David

closed 2026-02-19 by davidmpye