davidmpye/V10_Dyson_BMS #14: Failed to unlock

HellaK-png, 2024-06-19:
Hi David, I am trying to unlock my V10 pack with your firmware and a pi 2. After running I get this error.
[image]
"Unlock Failed"
"Unable to unlock pack"
Then it aborts.
Thanks.

davidmpye, 2024-06-19:
Hi,

It's most likely a connection issue - there is insulating varnish on the
pin-holes that needs to be carefully removed to ensure good contact.

David

HellaK-png, 2024-06-19:
Hmmm, I've resoldered all the connections and I'm still getting the error. The pack flashes 5 red lights when it is attempting/unable to unlock.
Thanks.

HellaK-png, 2024-06-21:
For anyone else with this issue, I followed another users comment and wired the 3.3v lead. This made the programming successful. 
Thanks for your work on this project David!

davidmpye, 2024-06-21:
Hi,

I'm really pleased it worked, but surprised that it needs the 3.3v supplied
externally.

I had advised against it as there's a small risk that the 3.3v regulator on
the board might backfeed the vReg on the Pi. Or potentially vice versa.

But if it works....!

David

davidmpye, 2025-01-12:
> I had to plug in 3.3v as well, mine was the 2300 MaH version if it matters.
>
> Also to anyone needing to remove the conformal layer - magic sponge works
> very well. If you're not sure what it is, it's a very lightly abrasive
> block of foam typically used in kitchens.

Interesting - I wonder if the packs are going straight back to sleep, so
the regulator isn't providing 3v3 from the pack cells.

Another thing you can use to remove the lacquer is either a soldering iron
(by tinning the pads but be careful not to short them out!), or a
fibreglass pencil like:

https://www.amazon.co.uk/Modelcraft-PBU1019-1-Glass-Pencil/dp/B002Z8IX36

David

BBBernsteyn, 2025-08-31:
Had the same issue before it was solved by connecting 3.3v as @HellaK-png suggested. Thanks @davidmpye for putting in the work _and_ sharing it with the wider world, thanks @HellaK-png for reporting the issue and providing a solution.

closed 2026-02-19 by davidmpye