davidmpye/V10_Dyson_BMS #16: Pi 5 support

bobfiggs, 2024-09-23:
Hi David,

Great project!

I have followed your flashing Wiki. I am using a Pi5 and I am getting an error that the installed operating system does not indicate support for Raspberry Pi 5.  
Any suggestions?

Thank you
Bob

davidmpye, 2024-09-23:
Hi Bob,

Where are you seeing this error?

I haven't built or tested with a pi5 so you are a bit ahead of the curve

David

bobfiggs, 2024-09-23:
David,

Thanks for the quick response.  

I get notification on boot up of the Pi. Please see attached.

I have tried editing the config file with the os_check=0 but then the pi does nothing on boot.  No video, the onboard LED just stays solid green. 

Bob

[image]

bobfiggs, 2024-09-30:
David,

Got a rpi zero and used v1.2, battery now works!

Thanks for sharing!

closed 2024-09-30 by bobfiggs