davidmpye/V10_Dyson_BMS #18: BMS Dead after flashing

coxtor, 2025-02-02:
Hi, thanks for this amazing Project! I tried the recent zip image with a raspberry pi and it appeard to have worked fine (flash sucessfull message on screen). I flashed with the old cells in place, replaced them, measured the voltage, all appeared fine. What was weird from the beginning: the battery indicator was always "low" when activating the battery, even though, the cells were all at 4.21V.  I inserted the battery in the vacuum and it worked great while testing (60 sec vaccuming and more with max suction compared to only 7 remaining seconds). 

When inserting it in the charger nothing happened. Then after letting it sit a while the bms now does nothing. I'm also no longer able to flash the device, as swd can no longer find the device.  Also I can't wake the battery any longer, it just does nothing. 

Any idea what happend and where to continue ? I'm not sure but I have the feeling that it is dead dead ?

I took loads of pictures during the progress. feel free to use them as you please. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Reddwarfusa, 2025-02-07:
I'm having the same issue,  After flashing the pack will not wake up.  Zero activity on the onboard LED.  All cells are 3.95 or above.  Is there a way to resurrect?

coxtor, 2025-02-15:
I managed to get it working! I bought a defective battery for 10€, so I gave it another try.

However, after assembling the kit again, the battery was low again, even though the pack was fully precharged (each cell had between 4.18 and 4.19 volts). I don't know if it's important, but I flashed the firmware again after replacing the batteries. That actually changed the indicator to "level 2". Now it's sitting in the Dyson charger and appears to be working as expected—wait and see!

Tips:

    Be really careful when soldering 18650 cells. After remeasuring the voltages of individual cells, it appears I killed two batteries in the pack.
    I must have shorted something during the initial flashing process, as my Pi is also dead.

Anyway, thanks so much for this project!
It probably would have been more time- and cost-efficient to just buy a new original pack from Dyson. I ended up buying 11 batteries (two extra as backups in case I killed more). But on the other hand, it was fun, I know exactly what I’ve got compared to third-party packs, and I'd like to make myself believe that I have a superior runtime compared to a stock battery.

Last but not least—I did something for the environment :) … apart from frying my own battery and having to recycle a defective one.

closed 2025-02-15 by coxtor