davidmpye/V10_Dyson_BMS #19: LED Flashing Light Sequence

kcollins7, 2025-02-28:
So I finished flashing my battery pack via my Pi 4 and everything seems to have gone well. Got a bunch of flashing red LEDs and waited a good while before disconnecting the wires I had soldered and cleaning up.

According to your guide the pack LEDs should flash in a particular sequence when it wakes up from sleep but I don't see any sign of that when I push the little button that turns on the battery status LEDs. Is this expected behavior?

Also may have briefly touched the 3.3V and GND pads with my soldering iron and the device briefly flashed but hopefully it didn't break anything. I haven't been able to test it yet but pushing the black button to turn on the battery level indicator still works fine. Let me know if there's anything I might want to look out for.

davidmpye, 2025-03-02:
Hi,

If the firmware has been flashed successfully, when the pack wakes from sleep, all the lights should briefly illuminate in sequence, up and down.

This only happens when the pack wakes from sleep - it takes about 5 mins to go to sleep from last use..  When you press the button after 5 mins you should see the light sequence once.  You won't see it after every press, only if the pack has been in sleep (not idle), if that makes sense.

I'm not sure why you saw red LEDs flashing at the time of the flash process - that seems odd.  Did the Pi report a successful flash?
Also, was your pack working prior to flashing the firmware, or was it locked out?  If it was locked out before, and now it operates the cleaner, the flash must have worked...     Otherwise, I'd be inclined to say, based on what you say, that it hasn't flashed, and is still running the old firmware.

David

kcollins7, 2025-03-03:
The Pi 4 did report a successful flashing and I finally saw the expected behavior today of a VERY quick light flash sequence when I pushed the button. My battery was not dead before I flashed but it would not last very long so that's why I started to look into replacing the 18650 cells since new replacement batteries are way overpriced. I'm guessing everything is working as intended. Do you know if there's a guide or information somewhere about replacing the batteries in this pack? Do people usually get new strips or reuse the old ones? Should the batteries be fully charge before installing them and spot welding again? Thanks!

kcollins7, 2025-03-05:
Got the new P28a batteries installed and everything seems to working great. Charger gets them to about 4.15V after I installed them at 3.7V and all the modes work as well. Cheers!